maximkulkin/esp-ir #1: Worked on Samsung TV Remote. Troubles with A/C

peros550, 2019-04-21:
Hi Maxim, 

today I found some time to test your library. The IR Dumper example worked out of the box. 
 
The IR Send code found in the repo's start page needed a small modification: 

`static int16_t[] command1 =.... >> static int16_t command1[] =....`

As mentioned, I was successful testing Samsung TV IR commands, but I wasn't successful testing A/C IR commands. 
If you have any advice on how to troubleshoot or tinker it, it would be greatly appreciated. 

Many Thanks!!!

maximkulkin, 2019-04-21:
Without any details I can only suggest to test harder... and try to localize the problem.

peros550, 2019-04-22:
Hi Maxim, 

I'm trying to reproduce an IR command of a Toyotomi A/C. 
I used the raw_dumper example and here is what a turn ON command at 30C heating gave:

```
Decoded packet (size = 139):
9021 ,-4457 ,  677 , -556 ,  652 , -554 ,  654 ,-1627 ,  683 ,-1625 ,  674 ,-1634 ,  676 ,-1630 ,  680 , -554 ,
654 , -552 ,  656 , -551 ,  656 ,-1625 ,  675 ,-1633 ,  677 ,-1630 ,  680 , -553 ,  655 , -551 ,  656 , -551 ,
657 , -549 ,  648 , -559 ,  649 , -558 ,  649 , -557 ,  651 , -555 ,  652 , -555 ,  653 ,-1628 ,  682 ,-1626 ,
674 , -558 ,  649 , -558 ,  650 , -556 ,  651 , -555 ,  653 , -553 ,  654 ,-1627 ,  684 , -549 ,  648 ,-1633 ,
677 , -556 ,  651 , -555 ,  653 ,-1627 ,  683 , -550 ,  647 ,-19965 ,  678 , -555 ,  653 ,-1629 ,  681 , -552 ,
656 , -551 ,  656 , -551 ,  657 , -550 ,  647 , -559 ,  648 , -558 ,  650 , -557 ,  651 , -555 ,  652 , -554 ,
654 , -552 ,  656 , -550 ,  647 ,-1634 ,  676 , -556 ,  652 , -555 ,  652 , -554 ,  654 , -552 ,  656 , -550 ,
657 , -550 ,  647 , -559 ,  649 , -557 ,  650 , -556 ,  652 , -554 ,  654 , -552 ,  655 , -552 ,  656 , -550 ,
647 , -559 ,  648 , -558 ,  650 ,-1631 ,  679 ,-1628 ,  682 , -551 ,  657 ,
```

Then  I used the following code to transmit the IR code using a Wemos D1 mini: 

```
#include <stdio.h>
#include <stdlib.h>
#include <esp/uart.h>
#include <FreeRTOS.h>
#include <task.h>

#include <ir/ir.h>
#include <ir/raw.h>

static int16_t command1[] = {9021 ,-4457 ,  677 , -556 ,  652 , -554 ,  654 ,-1627 ,  683 ,-1625 ,  674 ,-1634 ,  676 ,-1630 ,  680 , -554 ,
654 , -552 ,  656 , -551 ,  656 ,-1625 ,  675 ,-1633 ,  677 ,-1630 ,  680 , -553 ,  655 , -551 ,  656 , -551 ,
657 , -549 ,  648 , -559 ,  649 , -558 ,  649 , -557 ,  651 , -555 ,  652 , -555 ,  653 ,-1628 ,  682 ,-1626 ,
674 , -558 ,  649 , -558 ,  650 , -556 ,  651 , -555 ,  653 , -553 ,  654 ,-1627 ,  684 , -549 ,  648 ,-1633 ,
677 , -556 ,  651 , -555 ,  653 ,-1627 ,  683 , -550 ,  647 ,-19965 ,  678 , -555 ,  653 ,-1629 ,  681 , -552 ,
656 , -551 ,  656 , -551 ,  657 , -550 ,  647 , -559 ,  648 , -558 ,  650 , -557 ,  651 , -555 ,  652 , -554 ,
654 , -552 ,  656 , -550 ,  647 ,-1634 ,  676 , -556 ,  652 , -555 ,  652 , -554 ,  654 , -552 ,  656 , -550 ,
657 , -550 ,  647 , -559 ,  649 , -557 ,  650 , -556 ,  652 , -554 ,  654 , -552 ,  655 , -552 ,  656 , -550 ,
647 , -559 ,  648 , -558 ,  650 ,-1631 ,  679 ,-1628 ,  682 , -551 ,  657};

void user_init(void) {
    uart_set_baud(0, 115200);
	
	 printf("Starting \n");
	ir_tx_init();
	
	while(1)
	{
	 printf("Sending \n");
	ir_raw_send(command1, sizeof(command1) / sizeof(*command1));
	
	vTaskDelay( 1500 / portTICK_PERIOD_MS );
	}

}
```
 Having two Wemos D1 mini, I has able to send IR commands on one board and capture the command using the second board at the same time. Here is a picture of the setup in action. 

[image]

The raw dumper captured the following:

```
Decoded packet (size = 137):
9082 ,-4436 ,  722 , -512 ,  706 , -526 ,  703 ,-1603 ,  729 ,-1599 ,  722 ,-1591 ,  730 ,-1603 ,  729 , -528 ,
701 , -529 ,  700 , -504 ,  704 ,-1600 ,  732 ,-1604 ,  728 ,-1658 ,  674 , -587 ,  621 , -525 ,  704 , -525 ,
704 , -523 ,  706 , -507 ,  700 , -577 ,  663 , -598 ,  610 , -531 ,  708 , -525 ,  704 ,-1603 ,  729 ,-1600 ,
700 , -536 ,  704 , -532 ,  697 , -512 ,  707 , -526 ,  702 , -530 ,  699 ,-1581 ,  754 , -498 ,  707 ,-1605 ,
727 , -608 ,  621 , -511 ,  707 ,-1598 ,  734 , -519 ,  699 ,-20506 ,  700 ,-1607 ,  725 , -505 ,  724 , -556 ,
652 , -566 ,  663 , -526 ,  703 , -534 ,  684 , -527 ,  702 , -533 ,  696 , -512 ,  706 , -526 ,  703 , -527 ,
[... 34 lines cut ...]
700 , -511 ,  708 ,-1601 ,  730 ,-1602 ,  729 , -598 ,  631 ,
Decoded packet (size = 1):
206 ,
Decoded packet (size = 137):
8990 ,-4431 ,  724 , -598 ,  610 , -534 ,  705 ,-1600 ,  731 ,-1659 ,  662 ,-1589 ,  732 ,-1601 ,  730 , -527 ,
701 , -529 ,  700 , -529 ,  700 ,-1579 ,  731 ,-1669 ,  662 ,-1698 ,  633 , -532 ,  697 , -508 ,  752 , -477 ,
699 , -528 ,  701 , -597 ,  621 , -529 ,  699 , -536 ,  703 , -506 ,  702 , -531 ,  708 ,-1598 ,  733 ,-1596 ,
725 , -566 ,  652 , -529 ,  699 , -511 ,  697 , -536 ,  703 , -587 ,  642 ,-1604 ,  727 , -525 ,  672 ,-1615 ,
726 , -533 ,  706 , -502 ,  759 ,-1547 ,  721 , -532 ,  707 ,-20497 ,  703 ,-1606 ,  725 , -535 ,  694 , -529 ,
700 , -513 ,  694 , -566 ,  663 , -533 ,  675 , -608 ,  631 , -545 ,  694 , -526 ,  681 , -527 ,  702 , -528 ,
701 , -527 ,  701 ,-1586 ,  725 , -534 ,  705 , -528 ,  700 , -556 ,  652 , -529 ,  700 , -597 ,  631 , -588 ,
641 , -512 ,  706 , -529 ,  700 , -608 ,  599 , -534 ,  705 , -525 ,  704 , -526 ,  703 , -526 ,  681 , -531 ,
708 , -528 ,  680 ,-1604 ,  727 ,-1605 ,  726 , -528 ,  701 ,
Decoded packet (size = 1):
228 ,
```
 One quick observation has to do with the size. While the IR code sent by the original remote gave a size of 139 , the IR code reproduced by your library was captured as 137. 

Then I tried to compare the IR Codes using a program called AnalysIR. 

Here is the result: 

[image]

The yellow pulses come from the original remote while the green comes from the WemosD1 mini.

maximkulkin, 2019-04-22:
Oh, thank you for detailed description.

I've seen internal timer that is used to time pulses having issues with delays >= 10000, it will somehow consume the next pulse. When I first found it, I decided not to deal with it assuming protocols should not have such long pulses/silences. But it might be worth investigating in depth.

Can you try this: replace -20000 you have with two values of -9999 (absolute value should be less than 10000) and see if it helps.

peros550, 2019-04-22:
Thanks Maxim! I'll try and let you know!

peros550, 2019-04-22:
Maxim! Problem solved! Many Thanks!!!!!!

maximkulkin, 2019-04-22:
I will probably implement a workaround for that to internally split long pulses into multiple timer calls until I figure out what is the root cause. Feels like there is a bug in esp-open-rtos using FRC1 timer with 256 divider.

closed 2019-04-22 by maximkulkin

peros550, 2020-10-17:
> I will probably implement a workaround for that to internally split long pulses into multiple timer calls until I figure out what is the root cause. Feels like there is a bug in esp-open-rtos using FRC1 timer with 256 divider.

Hi Maxim, 

did you have time to check the issue with values greater than 9999 ?

peros550, 2020-10-29:
@maximkulkin hi Maxim, I hope you are doing well . Do you plan a fix for the above issue?