maximkulkin/esp-ir #2: Schematic circuit diagram of the IR part

peros550, 2019-04-24:
Hi Maxim, 

I wanted to ask you about the circuit diagram. I'm using the following schematic in regards to the IR part. 
[image]

Are you using the same or anything similar? Do you see any room for improvements or corrections?  

Thanks
Yiannis

maximkulkin, 2019-04-24:
Pretty much. I'm also experimenting with not having a resistor for LED (R2 on your schematic), since power supply I plan to use can not output high currents anyways.

peros550, 2019-04-24:
What about R1? Would you think a value around 220 Ohm would make any difference for better results ?

maximkulkin, 2019-04-24:
Ok, here are calculations: IR diodes I plan to use have forward voltage of 1.3V and forward current 30mA. I plan to power it with 3.3V. So R2 value should be (3.3 - 1.3) / 0.03 = 66.67 Om, which corresponds to standard value of 68 Om.
PN2222 has collector current rating of 600mA, so it is fine there. For 30mA collector current transistor currnet gain is somewhere around 50-60 (assume 50), so base current should be 30 / 50 = 600uA. Since transistors usually have base emitter voltage drop of ~0.6V and drive voltage is 3.3V, R1 = (3.3 - 0.6) / 0.0006 = 4500 Om. You can take something around 3.3kOm and should be fine. I've used 1kOm resistor.

PS I'm not an electronics expert, but calculations seem to be correct

peros550, 2019-04-25:
Thank you Maxim, once again 👍

closed 2019-04-25 by peros550