maximkulkin/esp-ir #3: Request: separate ir_mark and ir_space functions

Korcsog, 2019-06-06:
Hi!

Really great solution for sending IR codes, but I'm really missing the separate IR.mark and ir.Space functions. My AC sends 8 bits splitted in two parts and its hard to implement because I need these functions like in [ir-esp-uart](https://github.com/pbendersky/ir-esp-uart). The only problem with this lib, it uses GPIO2 for TX and this is also the built-in LED's GPIO.

Is it possible to implement these two functions using I2S?

maximkulkin, 2019-06-06:
As far as I understand, those functions are for starting and stopping to emit carrier signal. The problem with that approach is: it's too low level. You will have to manage timings and alternate between emitting and silence manually.

This library has notion of [encoders](https://github.com/maximkulkin/esp-ir/blob/master/ir/tx.h#L8-L14): it's an object that is passed to TX system and it tells TX what to do next. You just need to implement your own encoder function that will return period of next mark/silence (with positive value being mark and negative - silence). [Here](https://github.com/maximkulkin/esp-ir/blob/master/ir/raw.c#L10-L30) is implementation for "raw" TX where you provide all pulse length explicitly:
```c
typedef struct {
  ir_encoder_t encoder;

  size_t count;
  size_t pos;
  int16_t data[];
} ir_raw_encoder_t;

static int16_t ir_raw_get_next_pulse(ir_raw_encoder_t *encoder) {
    if (encoder->pos >= encoder->count)
        return 0;

    return encoder->data[encoder->pos++];
}
```
(there is a little bit more plumbing involved, but the essence is just the function and struct above)

If you do not want to implement your own encoder, you could just allocate an array of int16_t, fill it in with your pulses, call `ir_raw_send()`, free your array and be done with it. If you want to be more effective, you could implement a custom protocol like (e.g. like [this](https://github.com/maximkulkin/esp-ir/blob/master/ir/generic.c#L12-L99))

Korcsog, 2019-06-12:
@maximkulkin 

It sounds a little complicated for me, afraid of its beyond my capabilities 😅
Basically the ir_generic_send() function will working, but [my AC](https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Gree.cpp) send 3 messages in row. The first message contains the first 4 bytes, but this doesn't have footer mark/space. The second message contains only 3 bits and this one doesn't have header mark/space. The third one also doesn't have header mark/space, and contains the last 4 bytes of the message.
Can I manage the generic encoder to leave the footer/header mark space calls from the function itself?
Basic idea is to call `ir_generic_send()` three times.

Also if someone else could help... :

```
void IRsend::sendGree(unsigned char data[], uint16_t nbytes, uint16_t repeat) {
  if (nbytes < kGreeStateLength)
    return;  // Not enough bytes to send a proper message.

  for (uint16_t r = 0; r <= repeat; r++) {
    // Block #1
    sendGeneric(kGreeHdrMark, kGreeHdrSpace, kGreeBitMark, kGreeOneSpace,
                kGreeBitMark, kGreeZeroSpace, 0, 0,  // No Footer.
                data, 4, 38, false, 0, 50);
    // Footer #1
    sendGeneric(0, 0,  // No Header
                kGreeBitMark, kGreeOneSpace, kGreeBitMark, kGreeZeroSpace,
                kGreeBitMark, kGreeMsgSpace, 0b010, 3, 38, false, 0, 50);

    // Block #2
    sendGeneric(0, 0,  // No Header for Block #2
                kGreeBitMark, kGreeOneSpace, kGreeBitMark, kGreeZeroSpace,
                kGreeBitMark, kGreeMsgSpace, data + 4, nbytes - 4, 38, false, 0,
                50);
  }
}
```

maximkulkin, 2019-06-12:
This will not work (at least reliably) on RTOS because you need to be very precise with timings between calls. This worked on Arduino Core, because MCU does not do anything else, which is not the case on RTOS. I do not see reason to keep pushing to support this approach.

closed 2019-06-18 by Korcsog