maximkulkin/esp-ir #5: Doubt: about sending raw values

beckmx, 2019-07-30:
Hello Max,

I was looking at the demo, I noticed that my compiler refuses to compile with your example and decided to change it to:
`static int16_t command1[] = {}`
After that it compiles, anyhow, I dumped a ir remote with your library and gives me this:
`Decoded packet (size = 67):
 8954 -4567   490  -617   514  -595   486  -619   512 -1754   461  -645   486  -621   463  -617 
  489  -618   488 -1778   461 -1750   489 -1749   513  -620   486 -1727   513 -1751   486 -1727 
  488 -1751   487 -1751   512 -1752   462  -617   514  -619   462  -619   512  -594   512  -594 
  513 -1751   487  -619   488  -592   514 -1727   512 -1727   512 -1728   511 -1728   512 -1727 
  512  -619   488 `

I also dumped with the IRRemoteESP8266, just to compare reasons and gave me this:
`{8946, 4544,  538, 566,  542, 592,  512, 566,  538, 1724,  512, 568,  512, 594,  512, 594,  512, 594,  536, 1700,  538, 1700,  512, 1726,  512, 594,  510, 1726,  536, 1726,  486, 1724,  512, 1750,  542, 1670,  510, 1724,  514, 592,  512, 618,  486, 594,  512, 594,  510, 594,  512, 1752,  486, 594,  510, 568,  538, 1726,  510, 1722,  512, 1726,  510, 1724,  512, 1724,  510, 620,  460, 40564,  8946, 2326,  460};  // NEC 10EFC13E`

According to the example I should leave the - negatives? or should I remove them?, with the negatives like this:
`static int16_t command1[67] = {8954, -4567, 490,-617, 514,-595, 486,-619, 512 -1754, 461,-645, 486,-621, 463,-617,
489,-618, 488 -1778, 461 -1750, 489 -1749, 513,-620, 486 -1727, 513 -1751, 486 -1727,
488 -1751, 487 -1751, 512 -1752, 462,-617, 514,-619, 462,-619, 512,-594, 512,-594,
513 -1751, 487,-619, 488,-592, 514 -1727, 512 -1727, 512 -1728, 511 -1728, 512 -1727,512,-619, 488};

void ir_dump_task(void *arg) {
    while(1){
        ir_raw_send(command1, sizeof(command1) / sizeof(*command1));
        vTaskDelay(5000 / portTICK_PERIOD_MS);
    }

}

void user_init(void) {
    uart_set_baud(0, 115200);
    ir_tx_init();
    gpio_enable(10, GPIO_OUTPUT);
    gpio_write(10, 1);
    xTaskCreate(ir_dump_task, "IR dump", 2048, NULL, tskIDLE_PRIORITY, NULL);

}`
I cannot get it to transmit (it does blink the led) and my other doubt is:
` ir_raw_send(command1, sizeof(command1) / sizeof(*command1));` should it be only the sizeof(command1)?

Fonger, 2019-07-30:
You should leave the negative.  It stands for no pulse time span in esp-ir.
The other libraries(like IRRemoteESP8266 you mentioned) do the same thing.
However they distinguish them by odd/even item of the signal array, instead of positive/negative values.

I find the implementation by esp-ir useful because there's some trouble when FRC1 timer >= 10000us. I use multiple negative values in a row to address this as workaround.

beckmx, 2019-07-30:
btw I noticed the 40564 and according to the comments in other post, so I should use 4 values of -9999.

Yes looks like the counter is skipping the numbers afterwards, I will try that

Fonger, 2019-07-30:
Actually, your 40564 and the following 8946, 2326, 460 is repeat code in NEC protocol. The real signal ends before 40564us delay. It should work without them.

closed 2019-07-31 by beckmx