maximkulkin/esp-ir #9: Suggestion: Disable GPIO 14 pull up on init

Fonger, 2019-09-17:
TL;DR

I find that esp8266 GPIO14(D5) although it's not high on boot and after boot, but it is by default "pull-up" on boot and even after boot. So I suggest disable pull-up at user_init() to prevent IR led burn.

```c
void user_init(void) {
    uart_set_baud(0, 115200);
    gpio_set_pullup(14, false, false); // or gpio_write(14, false);
    //....
    //....
    ir_tx_init(); // or put inside this function, but I call this after homekit is paried. Maybe too late.
}
```

You may do some experiments yourself. I want to add a reference but I can't find it because most people cares about HIGH/LOW. GPIO14 D1 D2 is the most 
Just replace your IR led with visible light led and proper resistor.

You can find the LED light "slightly" even after boot. This is because GPIO 14 is pulled up by default and has small voltage.
[image]

If GPIO14 this is then connect to NPN transistor to switch on/off the IR led, then the IR led will be ON state whole time until you send any IR signal at lease first time! (because clr_carrier explicitly write that pin to low). I use 1k ohms resistor for my 2N2222 base and this GPIO 14 pull-up voltage is still high enough to switch on the IR led. (I replace IR led with Red LED for demostration)

[image]

## My Story behind this suggestion

I use very low resistor for my IR led because my esp8266 board is far away from my AC. It works total fine because IR led should only turn on for short period during signal transmission. And it's 50% duty cycle hence reduce effective current by half.

Yesterday I find I broke my IR LED. I moved my board to another place, hence the power to my NodeMCU board was temperately down and back thus the system reboot. Actually I had done this several times without any issues. The only difference is that: this time I don't test if it still works after boot so it doesn't send at least one IR led signal. And about one day later, I find it no longer works. I think the reason is that the IR led stay on with high current > 24 hours! I think this is the reason why it IR led burns. So I want to share this result and suggestion. Thanks for reading this far

Tested and reproducible on hardware:
- NodeMCU v3 (3 boards)

Tested and reproducible on firmware:
- esp-open-rtos
- ESP8266_RTOS_SDK v3.2

GPIO 14 is not HIGH, but pull up with small voltage by default during flash/boot/after boot.

maximkulkin, 2019-09-30:
It's good to know, but I do not think that this is something that should be handled inside this library (as you have mentioned that it happened prior to library initialization).
Also, this library's initialization is pretty lightweight, so you can do it on startup, no need to postpone it after homekit is paired.

closed 2019-09-30 by maximkulkin

peros550, 2019-10-01:
Thank you for the suggestion @Fonger !
I was doing many tests (trial and error)  with resistors , transistors in order to find the best combination that would provide reliable IR transmission. 

Sometimes the transistor run really hot right after a reboot. I think I may have burned one transistor.

Now I know why and included the function to my hardware init.   Perhaps it's not bad idea to include the above command in the initialization function.

Fonger, 2019-10-26:
@peros550 Yeah. Or at least show this caveat in README