maximkulkin/esp-ir #11: homekit and ir combination

andrei-gavrila, 2020-09-02:
Hi,

I'm having an intermittent issue with the following code. At random times, the IR sending part is no longer generating the proper signal to send the remote codes - sometimes are not recognised at all by the TV, sometimes are sent with minor variations and it is received as a different button press. I have tried to reinitialise the TX part by also changing the ir_tx_init function to delete the event group if defined, but I had absolutely no luck. Sometimes, the IR sending part comes back to proper behaviour, sometimes I loose my patience and I reset the ESP. Also, as you can see, I'm using ir_generic_send() and ir_raw_send(). Once sending is starting to randomly fail, the same applies to codes sent by the both functions. Can the timer be affected by the homekit part? As you can see from the code, the channel (input source) name is editable from the iPhone, so if you define an input source '51 Viasat History HD', the ESP will send the codes for 5, 1, OK to the TV when this input source is selected. Once the failure starts, I get random combinations of just 5 send, just 1 sent, just 5 and 1 sent, random button press sent - I even managed to turn off the TV several times :)

Here's the complete code:

```c
#include <stdlib.h>
#include <stdio.h>
#include <string.h>
#include <esp/uart.h>
#include <esp/gpio.h>
#include <espressif/esp_common.h>
#include <sysparam.h>
#include <FreeRTOS.h>
#include <task.h>
#include <homekit/characteristics.h>
#include <homekit/homekit.h>
#include <wifi_config.h>
#include <ir/ir.h>
#include <ir/generic.h>
#include <ir/raw.h>

homekit_server_config_t config;

#define CHANNELS 10

uint8_t ch_data[10][4] = 
{
    {0x20, 0x08, 0x47, 0xB8},
    {0x20, 0x08, 0x1C, 0xE3},
    {0x20, 0x08, 0x1D, 0xE2},
    {0x20, 0x08, 0x1E, 0xE1},
    {0x20, 0x08, 0x40, 0xBF},
    {0x20, 0x08, 0x41, 0xBE},
    {0x20, 0x08, 0x42, 0xBD},
    {0x20, 0x08, 0x44, 0xBB},
    {0x20, 0x08, 0x45, 0xBA},
    {0x20, 0x08, 0x46, 0xB9},
};

ir_generic_config_t protocol = {
    .header_mark = 9000,
    .header_space = -4500,

    .bit1_mark = 560,
    .bit1_space = -1690,

    .bit0_mark = 560,
    .bit0_space = -560,

    .footer_mark = 560,
    .footer_space = -9000,

    .tolerance = 20,
};

void ir_send(uint8_t *data, uint8_t size)
{
    ir_tx_init();

    ir_generic_send(&protocol, data, size);
}

struct
{
    uint8_t active;
    uint32_t active_identifier;
    char configured_name[1024];
    uint8_t sleep_discovery_mode;
[... 358 lines cut ...]
    }
    else
    {
        strncpy(device_state.configured_name, "Television", 1024);
    }
    device_state.sleep_discovery_mode = HOMEKIT_SLEEP_DISCOVERY_MODE_ALWAYS_DISCOVERABLE;

    homekit_service_t **srv = config.accessories[0]->services + 3;

    while (*srv)
    {
        if (sysparam_get_string((*srv)->characteristics[0]->value.string_value, &s) == SYSPARAM_OK)
        {
            homekit_value_destruct(&((*srv)->characteristics[2]->value));
            (*srv)->characteristics[2]->value = HOMEKIT_STRING(s);

            printf("user_init: set name %s for channel %s\n", s, (*srv)->characteristics[0]->value.string_value);
        }

        srv++;
    }

    wifi_config_init("Infrared Television Remote", NULL, on_wifi_ready);
}
```

peros550, 2020-11-01:
Hi, 

I was looking at your example above, I don't know if in the meanwhile you may have solved or not. 
I've noticed that you are calling the ir_tx_init(); in each send command. I did the same and had some similar issues with you. So I placed the ir_tx_init(); at my program's init function to make sure it's called just once and I think the code performed better.

maximkulkin, 2020-11-07:
@andrei-gavrila It would make sense to show your hardware setup and wiring.